cryptoadvance/specter-diy #293: Implement language selection in the settings

louisthecat86, 2025-02-28:
Hello everyone, 

I am an absolute novice in the field of programming. But I have realized that with the help of AI such projects could also be played through by beginners. 

My idea or my concern is to change the Specter menu, which is currently only available in English, so that there is a language selection and the menu can be translated into different languages accordingly. 

I would be very happy if you could help me with my project. As I said, I am an absolute newcomer and will therefore have a lot of questions. Nevertheless, I think that you can just get started even if you end up failing.

Translated with DeepL.com (free version)

Schnuartz, 2025-02-28:
Good idea, but this sounds like a bit work. @k9ert do you have any tips for him. When he tries to implement this himself?

k9ert, 2025-03-05:
This is definitely possible. Translating is called "i18n" which is "i", 18 letters and then "n": **Internationalisation**.
This is the right term if you conversations with AI. I myself am not sure yet how exactly approach this. In bigger software-projects, there is usually a framework or library used. The most famous one is "gettext".
We can't use this here. So yesterday i had this conversation with grok yesterday and we need something like this:

```python
# translations.py
translations = {
    "en": {
        "greeting": "Hello",
        "goodbye": "Goodbye",
        "status": "System is running"
    },
    "es": {
        "greeting": "Hola",
        "goodbye": "Adiós",
        "status": "El sistema está funcionando"
    },
    "fr": {
        "greeting": "Bonjour",
        "goodbye": "Au revoir",
        "status": "Le système fonctionne"
    }
}

# Global variable to track the current language
current_lang = "en"

def set_language(lang):
    global current_lang
    if lang in translations:
        current_lang = lang
    else:
        print(f"Language '{lang}' not supported. Falling back to 'en'.")
        current_lang = "en"

def t(key):
    """Translate a key into the current language."""
    return translations.get(current_lang, translations["en"]).get(key, f"[{key}]")

# Main program
if __name__ == "__main__":
    print(t("greeting"))  # Output: Hello
    set_language("es")
    print(t("greeting"))  # Output: Hola
    set_language("fr")
    print(t("goodbye"))   # Output: Au revoir
    print(t("unknown"))   # Output: [unknown] (fallback for missing key)
```

schnuartz-ai, 2026-08-30:
# Draft comment for issue #293 — "Implement language selection in the settings"

## Current assessment

**STILL RELEVANT (with a partial prototype in k9ert's fork, not merged).** The firmware UI is English-only: there is **no** i18n framework in `HEAD = 3f3c831` — `grep -rniE "def t\(|current_lang|translations\b|i18n"` across `src/` returns zero matches, and no `language` key is read from settings. A working prototype exists in k9ert's fork (commit `75196b633b8fd10dbd62399e38dd2fcf6cb8f7b8`, "feature: switch language", 2025-04-04) which adds a "Change Language" button and a `change_language()` method to `src/specter.py`, but it (a) is **not** in this repo's history, and (b) provides only the language *switcher* UI — it does **not** include the `t()` translation function or any translation files, which a collaborator (louisthecat86) was separately assembling (per issue #298 comments). The Schnuartz comment on this issue (2026-06-02) notes "Marko is working on that for the Specter 3.0 firmware", which I could not corroborate in-repo (no such branch/commit found; the "3.0 firmware" is not in this repo's history).

The scope is moderate but non-trivial: ~371 user-facing English string literals across `src/` (menu titles, button labels, prompts, alerts, transaction-screen text) would need wrapping, plus a translation dictionary mechanism (k9ert suggested a simple `translations` dict + `t(key)` in the #298 discussion; gettext is not usable on this micropython target). The more detailed design discussion lives in the sibling issue **#298 "Implementing a language button"**, which is the actionable duplicate of this one.

## What I investigated

- `gh issue view 293` — body asks for a language-selection menu so the English-only UI can be translated; author is a self-described beginner. Three comments: Schnuartz (2025-02-28) pings @k9ert for tips; k9ert (2025-03-05) explains i18n, says gettext can't be used here, sketches a `translations` dict + `current_lang` global + `t(key)` function, and offers a call; Schnuartz (2026-06-02) says "Marko is working on that for the Specter 3.0 firmware. And that it's still working." No labels.
- Searched current repo for any i18n work: `grep -rniE "def t\(|current_lang|translations\b|i18n|language"` across `src/*.py` → no matches (the only `language`/`LANGUAGES` hits are in the bundled embit BIP-39 wordlist code under `f469-disco/libs/common/embit/`, unrelated to UI translation). Confirmed the UI is English-only.
- Inspected k9ert's referenced prototype via the GitHub API (`repos/k9ert/specter-diy/commits/75196b6…`): it modifies only `src/specter.py` (+58/-3), adding `(789, "Change Language")` to the `update_devsettings` button list and an `async def change_language(self)` method that presents `("en","English")`/`("de","Deutsch")`, stores `settings["language"]` via `self.GLOBAL`/`save_settings`, and shows a confirmation alert. It does **not** define a `t()` function or translation tables.
- Confirmed the prototype is **not** in this repo: `git log --all -- src/specter.py` shows no "language"/"switch language" commit; `git branch -a` shows no i18n branch on `origin`. The fork commit's parent (`24d1137`) is also not in this repo's history.
- Located the exact integration point in current code: `Specter.update_devsettings` at `src/specter.py:564-618` builds the settings menu; the "Change PIN code" entry is at `:576` (k9ert's prototype inserts "Change Language" just before it). The settings persistence mechanism the prototype relies on is `self.GLOBAL` / `BaseApp.GLOBAL` / `self.save_settings` at `src/specter.py:560-562` and `:646-648`.
- Sized the translation scope: `grep -roE '"[A-Z][a-zA-Z ]{3,40}"' --include="*.py" src/` → ~371 candidate user-facing string literals. Sample hotspots: `src/gui/screens/settings.py` ("USB communication", "Developer mode", OFF/ON labels), `src/gui/screens/transaction.py` ("Fee:", "INPUTS", "OUTPUTS", "Show detailed information"), `src/gui/screens/mnemonic.py` ("Hint: click on any word above…"), `src/specter.py` ("Device settings", "Reboot", "Wipe the device", "Change PIN code", "About this device"), and alert/prompt titles across hosts (`src/hosts/sd.py` "Overwrite?"/"Success!").
- `gh issue list --search "i18n language translation"` → #293 and **#298 (OPEN)** "Implementing a language button" — the latter has the detailed design thread (k9ert's `t()` sketch, the `current_lang = self.GLOBAL.get("language","en")` key line, and louisthecat86's progress). `gh pr list --search "i18n language translation"` → none.
- `git log --all --grep="i18n\|translation\|language" -i` → no relevant commits (only the bootloader bump etc. matched "latest").

## Current code

- **Settings menu (integration point for the switcher):** `Specter.update_devsettings` at `src/specter.py:564-618`. Button list built at `:565-580`; "About this device" `(42, …)` at `:573`, "Change PIN code" `(777, …)` at `:576` (conditional on `hasattr(self.keystore,"lock")`), "Reboot" `(456, …)` at `:578`, "Wipe the device" `(123, …, 0x951E2D)` at `:579`. Dispatch loop at `:581-618` (`menuitem == 255` back at `:587`; `777` → `self.keystore.change_pin()` at `:608-610`; `42` → `self.show_about()` at `:611-613`; `1` → `self.communication_settings()` at `:614-615`).
- **Settings persistence (the prototype's storage):** `self.GLOBAL` / `BaseApp.GLOBAL` assigned and saved at `src/specter.py:560-562` (in the experimental-settings setter) and loaded at `:646-648` (in `unlock`). `class Specter` declar [... cut ...] tion tables, and it is not merged into this repo. The actionable design thread is the sibling issue #298. Implementation: add a `src/i18n.py` with `translations`/`t()`/`set_language` (k9ert's #298 sketch), seed `current_lang` from `self.GLOBAL.get("language","en")` in `Specter.unlock` (`src/specter.py:647`), port `change_language()` + the `(789,…)` menu entry into `update_devsettings` (`:575`/`:608`), and wrap the ~371 user-facing literals starting with the settings/transaction/mnemonic screens.