cryptoadvance/specter-diy #298: Implementing a language button

k9ert, 2025-03-20:
Potentially, we could translate all the language in the DIY. But how would we switch between the languages?
We need to modify the GUI. Where do we want to add the functionality? Probably here:

[image]

Maybe before the "Change PIN code". So searching for the "Global Settings" reveals the function where this is located: `update_devsettings()` . It's only about 50 lines long and pretty straightforward. The buttons arrax needs to be **extend**ed at the correct position. One needs to specify a number (the existing number are a bit strange but they don't have any meaning, just choose a "random" one) and check later in the loop (`while True:`) for that number and call a new (async) function in that case. Don't forget the `await` if calling async functions.

The new function need to specify a button per language (sorry, no idea how some sort of dropdown can be used) which can be guesses looking at the `update_devsettings()` function.

So now, what to do if the user chooses one of the supported languages?

Schnuartz, 2025-04-03:
So I would handle that like the switching network. You press language and then there are a few languages you can choose. Funktionality would make sense in the settings tab right above the change PIN Code.

k9ert, 2025-04-07:
CC @louisthecat84

louisthecat86, 2025-04-08:
[image]

[image]

The first step is done but the next step is not really clear to me and I still have no solution how to get functionality on these buttons

louisthecat86, 2025-04-09:
But I tried your branch; it works great, and when you press the language selection button, you get a reaction by confirming and saving the selection. Now we have the facts: we have an i18n line of code that enables translation via the t() function, we have the translation files, at least I've collected all the terms that need to be translated, and we have the button function. Now we just need to put all these puzzle pieces together. I've already made a few attempts, but haven't had any success yet.

k9ert, 2025-04-10:
I think this is the key line:
https://github.com/k9ert/specter-diy/commit/75196b633b8fd10dbd62399e38dd2fcf6cb8f7b8#diff-a5ed4214bb2e2758cb52ae163779379af869e74b7e7782e1c56fb3f3eec7c268R558
```
current_lang = self.GLOBAL.get("language", "en")
```
You need to know what the current language is in order to decide in which language to translate the string passed from that function. 
So now the issue is that "self". It's the specter-object and whoever/whatever needs to know about the current language needs to have a reference to that specter object.
So now it depends on where we have that t-function. So the t-function needs to have a reference. I'm not sure where we have implemented the t-function, though.

schnuartz-ai, 2026-08-30:
# Draft comment for issue #298 — "Implementing a language button"

## Current assessment

**Feature request — still valuable, but not implemented.** This issue is k9ert's implementation plan for adding a UI language switch to the device-settings menu. No internationalization infrastructure exists in the codebase today: there is no `t()` translation function, no translation catalog/files, and no `language` setting in `GLOBAL`. The settings/GUI scaffolding the issue references (`update_devsettings()`) is present and matches the description, so the plan is still actionable, but none of the pieces it describes wiring together ("the `t()` function", "translation files") currently exist in this repo. This issue is effectively a more detailed duplicate of #293 ("Implement language selection in the settings"), which is the original feature request from the same contributor group.

## What I investigated

- `update_devsettings()` — the exact function the issue names as the place to add the button — its current button array and dispatch loop.
- The global-settings persistence layer (`GLOBAL`, `load_settings`/`save_settings`) to see how a `language` value could be stored.
- Any existing i18n / translation / `t()` / locale infrastructure across `src/`, `manifests/`, and all git branches.
- The `BaseApp` class (`can_process`/`get_prefix`) since the issue discusses the `self`/`GLOBAL` reference needed by a translation function.
- The k9ert branch/commit referenced in the thread (`75196b6`) to see whether any of it landed.
- Related issues via `gh issue list --search "language"`.

## Current code

The function the issue points at exists and matches the description. `src/specter.py:564-618`:

```python
async def update_devsettings(self):
    buttons = [
        (None, "Categories")
    ] + [
        (1, "Communication"),
        # (2, "Applications"),
        # (3, "Experimental"),
    ] + [
        (None, "Global settings"),
        (42, "About this device"),
    ]
    if hasattr(self.keystore, "lock"):
        buttons.extend([(777, "Change PIN code")])
    buttons += [
        (456, "Reboot"),
        (123, "Wipe the device", True, 0x951E2D),
    ]
    while True:
        menuitem = await self.gui.menu(buttons, ...)
        if menuitem == 255:
            return
        elif menuitem == 3:
            await self.experimental_settings()
        ...
```

The "Change PIN code" button the issue suggests inserting the language button before is at `src/specter.py:576` (`(777, "Change PIN code")`). The button-number/dispatch pattern is exactly as the issue describes: pick an arbitrary int, extend the `buttons` array, then add an `elif menuitem == <n>` branch in the `while True:` loop calling a new async function.

Settings persistence is in place and would host a `language` key:
- `src/specter.py:43` — `GLOBAL = {}` (class-level default).
- `src/specter.py:515-524` — `load_settings()` reads AEAD-encrypted JSON from `SETTINGS_DIR/global.settings`.
- `src/specter.py:526-533` — `save_settings()` writes it back.
- `src/specter.py:555-562` and `646-648` — the `experimental_settings` flow shows the pattern: build a dict, set `self.GLOBAL = settings`, `BaseApp.GLOBAL = settings`, then `save_settings(settings)`.
- `src/app.py:19` — `BaseApp.GLOBAL = {}` is the class attribute the issue's k9ert comment refers to ("whoever needs to know the current language needs a reference to the specter object"). Apps read settings via `BaseApp.GLOBAL`.

**What is missing** (verified by grep across `src/`, `manifests/`, and all branches):
- No `def t(` translation function anywhere.
- No `i18n` / `gettext` / `locale` / `translation` module.
- No translation files (no `en.json`/`de.json`/`.po`/`.yaml` catalogs in the tree).
- No `language` key referenced anywhere in `src/`.
- The k9ert branch line quoted in the thread — `current_lang = self.GLOBAL.get("language", "en")` — does not exist on `master` or any fetched branch.

So the premise in the issue body ("we have an i18n line of code that enables translation via the `t()` function, we have the translation files") does not hold for the current repo; that work lives only on the contributor's unmerged local branch.

## Reproduction / validation

- Static inspection only (no hardware). Confirmed by code search that the i18n prerequisites the issue says are "already there" are absent from the repository, so the feature cannot be wired up without first introducing the `t()` function and catalogs.
- Confirmed `update_devsettings()` location and the button-array/dispatch structure match the issue's guidance.

## Related issues/PRs

- **#293** (OPEN) — "Implement language selection in the settings" (opened 2025-02-28 by @louisthecat86). This is the original request; #298 is k9ert's maintainer follow-up with the implementation sketch. They are the same feature; #298 should be treated as a duplicate / detail companion to #293.
- #312 (OPEN) — "Proposal for more user centric UI/menu structure" — broader UI restructure that would interact with where a language selector lands.
- The k9ert commit referenced in-thread (`75196b6` on `k9ert/specter-diy`) is not merged and not present in any fetched remote branch.

## Implementation notes

If someone picks this up, the concrete steps against the current code:

1. **Add i18n plumbing (prerequisite, currently absent).** Introduce a `t(s: str) -> str` translation function that reads the active language from `BaseApp.GLOBAL.get("language", "en")` and looks up `s`  [... cut ...]
Feature request — still valuable, but currently unimplemented and a duplicate of #293. The settings/GUI hooks the issue references are present and the plan is sound, but the i18n prerequisites it claims already exist (the `t()` function and translation files) are not in the repository, so the issue's premise is partly outdated. Suggest consolidating with #293 and, if pursuing, scoping it as two phases: (1) land the `t()` infrastructure + settings key + language button with a single language, then (2) phase the string-by-string translation work.