cryptoadvance/specter-diy #302: SIGHASH_ALL flag appended to Schnorr signatures

jp1ac4, 2025-05-14:
It appears that Specter DIY is appending an explicit SIGHASH_ALL flag to Schnorr signatures for Taproot keypath and scriptpath spends, but I believe it can be omitted in this case as SIGHASH_ALL is the default value that will be assumed if no explicit sighash flag is appended.

jp1ac4, 2025-05-14:
cc @pythcoiner

schnuartz-ai, 2026-08-30:
**Current assessment**

STILL RELEVANT. The Specter DIY signing flow appends an explicit `0x01` (SIGHASH_ALL) byte to every Schnorr/Taproot signature it produces, even for ordinary key-path and script-path spends where BIP-341 says the sighash byte may be omitted (SIGHASH_DEFAULT, value `0x00`). The behavior is confirmed in the device's primary signing path (`sign_psbtview` → `Wallet.sign_input`/`keystore.sign_input` → `PSBTView.sign_input` → `sign_input_with_tapkey`). The produced signatures are still valid and consensus-legal, so this is not a fund-loss bug, but the issue's claim is correct and the fix is small. Verified statically against the vendored embit at `f469-disco/libs/common/embit/`; not reproduced by an on-device sign run in this audit.

**What I investigated**

- Fetched issue #302 (title "SIGHASH_ALL flag appended to Schnorr signatures", opened 2025-05-14, no labels, OPEN). Body: Specter DIY appends an explicit SIGHASH_ALL flag to Schnorr signatures for Taproot key-path and script-path spends, but it can be omitted since SIGHASH_ALL is the default assumed when no sighash flag is appended. One comment (`@jp1ac4`, 2025-05-14) pinging `@pythcoiner`. No PR or follow-up linked.
- Traced the device signing path end-to-end and confirmed the issue. Tracing below.
- `git log --all --grep="sighash" -i` → only `779c8d4 support sighashes and prvate keys in wallet` (2021). No commit addresses Taproot sighash-default handling. `git log --all -- f469-disco/libs/common/embit/psbtview.py` → empty (submodule, not tracked here directly). No relevant change in `src/apps/wallets/manager.py` history.
- Related issues: `gh issue list --search "sighash"` → #302 (this), #277 ("PSBT_TAP_KEY_SIG not supported"), #234, #283, #290, #293. #277 is the closest: it concerns the same `sign_input_with_tapkey` function and the same vendored embit. An existing audit comment on #277 already observes that the guard `if sighash != SIGHASH.DEFAULT: wit += bytes([sighash])` at `psbtview.py:652-653` is "correct" — but that guard only helps if `SIGHASH.DEFAULT` is actually *passed in*, which (as shown below) the device flow does not do for normal Taproot inputs. So #302 and #277 are complementary, not duplicates.

**Current code**

The default-sighash plumbing from the manager down to the Schnorr signer:

1. `src/apps/wallets/manager.py:393-433` — `confirm_sighashes`: for a normal PSBT (no custom sighash flags) it returns `None` (line 407). `confirm_transaction` then yields `options = dict(sighash=None)` (line 357).
2. `src/apps/wallets/manager.py:772-799` — `sign_psbtview(..., sighash)`. Line 784:
   ```python
   inp_sighash = sighash or inp.sighash_type or self.DEFAULT_SIGHASH
   ```
   With `sighash=None` and a Taproot input that carries no explicit `sighash_type`, this evaluates to `None or None or SIGHASH.ALL` → `SIGHASH.ALL` (= `0x01`).
3. `src/apps/wallets/manager.py:64` — `DEFAULT_SIGHASH = SIGHASH.ALL`. This is the root cause for the non-Liquid wallet manager.
4. `src/apps/wallets/wallet.py:276-290` — `Wallet.sign_input(..., sighash=SIGHASH.ALL, ...)`: forwards `sighash` straight into `psbtv.sign_input(i, k.private_key, sig_stream, sighash, ...)`.
5. `f469-disco/libs/common/embit/psbtview.py:686-804` — `PSBTView.sign_input`. Lines 722-732 map `SIGHASH.DEFAULT`→`SIGHASH.ALL` only for *non-taproot* inputs, then lines 780-796 call `sign_input_with_tapkey(..., sighash=inp_sighash)` with `inp_sighash = SIGHASH.ALL`.
6. `f469-disco/libs/common/embit/psbtview.py:631-657` — `sign_input_with_tapkey`, the key-path branch (lines 650-653):
   ```python
   sig = pk.schnorr_sign(h)
   wit = sig.serialize()
   if sighash != SIGHASH.DEFAULT:
       wit += bytes([sighash])
   ```
   Since `sighash == SIGHASH.ALL != SIGHASH.DEFAULT`, the `0x01` byte is always appended.
7. `f469-disco/libs/common/embit/psbtview.py:674-682` — script-path branch (lines 678-681):
   ```python
   sigdata = sig.serialize()
   # append sighash if necessary
   if sighash != SIGHASH.DEFAULT:
       sigdata += bytes([sighash])
   ```
   Same behavior for script-path spends.
8. `f469-disco/libs/common/embit/transaction.py:14-19` confirms `SIGHASH.DEFAULT = const(0)` and `SIGHASH.ALL = const(1)`.

Edge case also verified: even if a coordinator explicitly sets `sighash_type = 0` (DEFAULT) in the PSBT input, `psbtview.sign_input` reads `inp.sighash_type = 0`, but `None or 0 or SIGHASH.ALL` in step 2 treats `0` as falsy and still falls through to `SIGHASH.ALL` (verified in a Python REPL: `None or 0 or 1 == 1`). So the device can never emit a DEFAULT (no-byte) Taproot signature through this path.

The non-view path `Wallet.sign_psbt` → `psbt.sign_with(root, sighash=SIGHASH.DEFAULT)` (`psbt.py:919`) defaults to `SIGHASH.DEFAULT` and would behave correctly, but it is not the path used by the device signing flow (`sign_psbtview` is).

**Reproduction / validation**

- VERIFIED by static tracing (not by an on-device run): the chain `manager.sign_psbtview` → `Wallet.sign_input` / `keystore.sign_input` → `psbtview.sign_input` → `sign_input_with_tapkey` is the device's active signing path, and at every hop the effective sighash for an ordinary Taproot input resolves to `SIGHASH.ALL`.
- VERIFIED: the `if sighash != SIGHASH.DEFAULT` guard exists at `psbtview.py:652-653` and `psbtview.py:680-681`, so the *embit* layer is already capable of producing DEFAULT-style (byte-less) Schnorr sigs — the defect is purely that the device never passes `SIGHASH.DEFAULT` into it.
[... cut ...] hich is a fidelity/interoperability gap, not a vulnerability. No new private security detail.

Recommendation: STILL RELEVANT — confirmed in the device signing path (`src/apps/wallets/manager.py:784` resolving to `SIGHASH.ALL` for Taproot) and in `sign_input_with_tapkey` (`f469-disco/libs/common/embit/psbtview.py:652-653, 680-681`). The fix is small and local to the manager's sighash resolution; embit already handles DEFAULT correctly. Coordinate with #277 if the embit bump is being done anyway.

Schnuartz, 2026-11-09:
Current Workaround with Sparrow:

Change SigHash from Default into All:

[image]

[image]

pythcoiner, 2026-11-10:
> Current Workaround with Sparrow:

Change SigHash from Default into All:

sorry but this is a stupid advice, you should just fix this very long standing bug, that is terrible UX and privacy wise